citbrains/OpenPlatform #4: ARMでのロボカップソースコードのコンパイル

yasuohayashibara, 2016-12-31:
@yotaseki @ChisatoKasebayashi 
Raspberry PIでロボカップのソースコードをビルドしていますが，hr42はすんなりとビルドできたのですが，認識ライブラリの方は以下のエラーが発生しています．
Raspberry PIでRaspbianを使用しているという違いはありますが，Jestonと近い環境だと思いますので，心あたりがありましたら教えてください．

ちなみに，Ubuntuと同様のセットアップをしています．

```
cit@raspberrypi:~/citbrains_humanoid/for2050/src $ make
[  1%] Building CXX object HCServer/CMakeFiles/HCLib.dir/__/gen/StringSender.o
*** Error in `/usr/bin/c++': double free or corruption (top): 0x00f27398 ***
Aborted
HCServer/CMakeFiles/HCLib.dir/build.make:69: recipe for target 'HCServer/CMakeFiles/HCLib.dir/__/gen/StringSender.o' failed
make[2]: *** [HCServer/CMakeFiles/HCLib.dir/__/gen/StringSender.o] Error 134
CMakeFiles/Makefile2:93: recipe for target 'HCServer/CMakeFiles/HCLib.dir/all' failed
make[1]: *** [HCServer/CMakeFiles/HCLib.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

yotaseki, 2017-01-01:
CMakeList.txtに書かれている以下の行が原因です
```c
SET (CMAKE_CXX_FLAGS "-march=native")
```
私も詳しく把握しているわけではありませんが,ARMの場合はこの最適化オプションは使用できないみたいです.

yasuohayashibara, 2017-01-02:
@yotaseki ありがとう．ビルドできるところまで確認しました．

yasuohayashibara, 2017-01-02:
解決したので閉じます．

closed 2017-01-02 by yasuohayashibara

yasuohayashibara, 2017-01-03:
Raspberry Pi camera 2.1で画像処理部の動作確認ができた．
解像度はまだ640x480であるが，快適に動作している．
https://youtu.be/2m3d11I4ybo