citbrains/OpenPlatform #5: 無線LANがつながらない

yasuohayashibara, 2017-01-12:
WIKIに書いた設定を行っても無線LANが接続されない．
sudo iwlist wlan0 scan
ではアドレスが表示されるが，GUIで接続しようとしても接続できない．
手順を再度確認する．

ryuichiueda, 2017-01-12:
ラズパイ3でしょうか？Wikiまで追っかけられないんですけど、ラズパイの場合だと色々気をつけることがあります。

yasuohayashibara, 2017-01-13:
ラズパイ３です．
networkmanagerを入れると無線LANに繋がりましたが，動きが全体に緩慢になったような気がします．
（気がするだけかもしれませんが．．．）
ちなみに，VNCで操作しています．
あとで相談させて下さい．

ryuichiueda, 2017-01-13:
承知しました。ここらへんはノウハウの塊です。

yasuohayashibara, 2017-01-14:
後でwikiにまとめようかと思いますが，繋がりました．
自宅がまだWEPでしたので，それも問題を複雑にしていたかもしれません．

-----
WIFIの設定
http://qiita.com/yosi-q/items/c677e4650b22ffffa806

WEPの設定
http://nlogn.ath.cx/archives/001378.html

interfacesに以下を追加
wireless-power off

-----
解決したので閉じます．

yasuohayashibara, 2017-01-14:
そういえばnetwork-managerをapt-get removeしたら動きが早くなりました．
Jessieにnetworkmanagerは相性が悪いようです．

closed 2017-01-14 by yasuohayashibara