citbrains/OpenPlatform #6: RaspiberryPiのUARTのボーレートを上げる方法

yoshiya01, 2017-01-14:
藤田くん調査頼みました。(急がなくていいです)
現状クロックに影響されている？みたい
現状115200bpsであれば通信出来ますが、
1.5Mbpsでは出来てないのでこれを解決して下さい。
取り敢えず1.5Mbpsまで上げなくてもいいのでより高いボーレートで出来るようにしてみて下さい
(UARTのボーレートはいくつかパターンが有ります。)

バタバタしてて環境を整えきれてないので、
PCとRaspi間でのテストをお願いします。
これなら必要な物が少ないので。
必要なもの
UARTUSB変換(PC-Raspi間でテスト用)

ボーレートを上げられた様なら以下の物でテストして下さい。
メイン基板(RS485変換)
サーボ
サーボ書き換えツール(サーボ側のボーレートを書き換えるため)
電源

yasuohayashibara, 2017-01-14:
Raspi3ではないですが，すでにLinuxから1.5MBPSで通信できています．
手順をどこかに書いたような気がしますが，かなり特殊な処理をしないといけません．
（見つけたらこちらにリンク先を添付してくれると助かります．）

ソースコードはこちら
https://github.com/citbrains/citbrains_humanoid/pull/694

yasuohayashibara, 2017-01-24:
https://github.com/citbrains/citbrains_humanoid/commit/8126356cec6a4decc471b20caf04ec4b536c57a1
で対応しました．

以下，ラズベリーパイ＋自作回路から歩行パターンを再生している様子です．
https://youtu.be/Nf5K_VBkBDM

一つ注意事項として，
WiringPIを使用しましたが，従来のGPIO18がWiringPIだとGPIO1に割り当てられており，そこで時間を費やしてしまいました．今後使用する場合はお気をつけ下さい．

解決したので閉じます．

closed 2017-01-24 by yasuohayashibara