citbrains/OpenPlatform #7: v3基板改良点

yoshiya01, 2017-01-22:
ダイオード追加
Accelit_SWにLED(RS485切り替え部分のステータス表示)

yasuohayashibara, 2017-01-23:
「RS485切り替え部分のステータス表示」はあった方が良いですが，優先順位は低いです．
次の基板で欲しいのは，以下の機能です．
（外付けの基板でも構いません）

１）ステータスを表示するLED　(できれば4つぐらい)
２）音声出力（うるさい環境で使用するため出力大きめ）
３）音声入力

以上，ご検討下さい．

yasuohayashibara, 2017-01-25:
「リアルタイムクロック」も追加した方が良いです．

１）ステータスを表示するLED　(できれば4つぐらい)
２）音声出力（うるさい環境で使用するため出力大きめ）
３）音声入力
４）リアルタイムクロック

yasuohayashibara, 2018-10-27:
改良した基板
動作確認後にmasterに反映します．

[181026部品.xlsx](https://github.com/citbrains/OpenPlatform/files/2521271/181026.xlsx)
[101826_open_platfom.zip](https://github.com/citbrains/OpenPlatform/files/2521272/101826_open_platfom.zip)

yasuohayashibara, 2018-10-31:
動作確認のため製作して電源投入した．
まだ電源のみの確認

パターンにミスがあった（赤線）とはんだ付け時にパターンを損傷（黒色）してジャンパーピンをつけた．
[image]

電源は問題なく動作
https://youtu.be/yW3xjvKYTS4

以下，問題
１）0.1uFのコンデンサのはんだ付けが困難なため変更
２）抵抗が必要以上に小さいと思われるので，もう少し大きなものに変更したほうが良い．
３）ファンを固定するネジを長くする．今はM2.6x10