citbrains/OpenPlatform #23: キック力が弱い

yasuohayashibara, 2017-08-12:
ロボカップでシュートするのに対して，足先の先端速度が低い．

yasuohayashibara, 2017-08-12:
従来のKONDOと比べておよそ半分の速度になっている．
ちなみに，Dynamo (Acceliteの前) のボール速度は 2.5m/s という記録がある．

■ KONDO B3M-SC-1170-A
モータの最大回転数 46 rpm
4.81 rad/s
2.40 rad/s

股から足先 0.2	m
0.481 m/s
膝から足先 0.1	m
0.240 m/s

足先の速度
0.722 m/s

■ FUTABA RS405CB
モータの最大回転数 45 rpm
4.71 rad/s
股から足先 0.22 m
1.036 m/s
膝から足先 0.11 m
0.518 m/s

足先の速度
1.555 m/s

yasuohayashibara, 2017-08-12:
上記計算のスプレッドシート
[170813足先速度の計算.xlsx](https://github.com/citbrains/OpenPlatform/files/1220089/170813.xlsx)

yasuohayashibara, 2017-11-27:
実ロボットにおける検証

世界大会で開発したシュート
https://youtu.be/hlHyPo2hQXU
https://youtu.be/tprLfQ_ZzIM

支持脚のヨー軸を積極的に動かすシュートモーション
https://youtu.be/x_8HCoAuvjk
※飛距離を重視したため，安定性は無視している．

３ｍ程度安定してシュートできている．

yasuohayashibara, 2017-11-27:
意見があればお願いします．

例えば，
こちらもっと良い方法がある．
３ｍでシュート距離は十分である．
ハードウェアの改良が必要である．
など

SatoshiShimada, 2017-11-28:
シュート距離は3mあれば十分だと思います。
安定性を無視しているとありますが、手で支えなくてもキックできるのでしょうか？(動画からは支えているか判断できませんでした。)

takayan660, 2017-11-28:
手で支えてあげないと、キックする前に体勢を崩します
ですので、キック力はどこまでいけるか確認できたので、これからは体勢を崩さないようにできるキックモーションを考察する予定です

yasuohayashibara, 2017-12-16:
1.5mボールを転がすシュートを蹴れるようになったので，issueを閉じます．

closed 2017-12-16 by yasuohayashibara