citbrains/OpenPlatform #26: 入手性の良い広角カメラの検討

yasuohayashibara, 2017-11-18:
以下に入手性の良い広角カメラを見つけた．
https://www.switch-science.com/catalog/3211/

入手性や価格で有利であるため，こちらを採用することを検討する．
ちなみに，TX1にも接続できそうな気がする．
(議論されているのはRaspberry Camera V2ですが，おそらく互換性はあると思います．)
https://devtalk.nvidia.com/default/topic/972364/raspberry-camera-v2-Raspberry Camera-csi-with-jetson-tx1-j100-carrier-board/?offset=6

yasuohayashibara, 2017-11-20:
１個購入しました．

yasuohayashibara, 2017-11-27:
カメラ取り付けの様子（両面テープで貼り付けただけです）
[image]

撮影した画像（足元にボールを置き，カメラを下向きに傾けました．場所はキックオフの位置）
[image]

最大解像度2592 x 1944でキャプチャした画像の画面をキャプチャしたので，解像度は適当

結果的に，足元のボールもゴールラインのボールも両方視野内に入っている．
おそらく現在使用しているカメラでの画像とそれほど違わないと思われる．

yasuohayashibara, 2017-11-27:
OpenPlatformではこちらのカメラをまずは採用したいと思いますが，意見がありましたらお願いします．

SatoshiShimada, 2017-11-28:
ケーブルの取り回しが悪いため実戦投入が心配ですがどのようにお考えでしょうか。

yasuohayashibara, 2017-11-29:
実験はしていませんが，フラットケーブルの途中を割いてまとめると屈曲に対する耐性が上がるとの情報がありました．

あとは，最悪途中に変換基板を入れて対処することもできると思いますが，こちらが手間が多いですね．
フラットケーブル　→　基板　→　シリコンケーブル　→　基板　→　フラットケーブル

ちなみに，何もせずそのままで市販しているロボットもあります．
http://www.botmag.com/raspberry-pi-camera-robot/

SatoshiShimada, 2017-11-29:
割いて使用するのは信頼性が薄いので不安が残ります。
基盤を途中にいれるのは良いアイデアだと思いますが、あまり新規開発の案件を増やしたくないため、より一般的なUSBカメラでもよいのではないでしょうか。
#29 でチルト軸を扱うようになっているので広角カメラである必要もないのではないかと考えます。

yasuohayashibara, 2017-11-30:
意見ありがとう．
多方面からの意見がシステムの完成度を上げると思いますので，議論していきましょう．

基板にMIPI CSI-2のコネクタがあるので，試してみたいというのが一番ですが，実際にどちらが良いかは動かしてみて判断するのでも良いかと思います．
ケーブルに関しては，あまり検証はできていませんが，割かなくても動作することは動作します．
まずは，問題が発生するまで動かすことが重要かなと考えています．

USBに関してですが，複数のデバイスから多くのデータを流すとシステムが実時間で動かなくなることを経験していますので，その点でMIPIを採用した時にどのような変化があるかというのを検証したいと考えています．
実はこの件で一番影響を受けるのは歩行の安定性ではないかと考えています．

広角カメラに関してですが，狭角のカメラを当初は採用することを考えていましたが，現在の画像処理および戦略の開発リソースを見る限り，チルトも含めた画像処理システムを作り直すことは難しいと思いますので，まずは現在のカメラと同じ広角カメラで構成したほうが良いと考えています．

とにかく作って動かしてデータを取ることが必要だと考えています．

SatoshiShimada, 2017-11-30:
ありがとうございます。確かに安定して動作するなら採用したいですしそのためにも試しておくべきですね。

yasuohayashibara, 2017-12-11:
議論が一度収束したので，issueを閉じます．
再度議論する必要があれば，オープンします．

closed 2017-12-11 by yasuohayashibara