citbrains/OpenPlatform #28: 腕の長さの検討

yasuohayashibara, 2017-11-19:
腕の長さが足りないために起き上がりをしづらい．
適正な腕の長さを求める．

yasuohayashibara, 2017-11-19:
現在の腕の長さ
[image]

おそらく腕の長さを制限するロボカップのルールは以下となる．
• The robot does not possess a conﬁguration where it is extended longer than 1.5·Htop .

身長　　　608mm
腕の長さ　246.5mm
肩幅　　　183mm
で，(608 * 1.5 - 183 - 2 * 246.5) / 2 = 118

片腕118mm延長してもルール上は問題ない．

yasuohayashibara, 2017-11-19:
一方起き上がりに必要な腕の長さの検討
60mm程度伸ばせば起き上がりできると考える．

[image]

[image]

yasuohayashibara, 2017-11-19:
肩から肘までを40mm，肘から手先までを30mm伸ばして，合計70mm伸ばしたときの様子

[image]

この程度の長さの余裕があれば，問題なく起き上がりできると考える．

yasuohayashibara, 2017-11-19:
検討してみました．
抜けがあればご指摘ください．

yasuohayashibara, 2017-11-19:
直立したときの外観は以下のようになる．
（腕をまっすぐ伸ばすと干渉します．）

[image]

takayan660, 2017-11-19:
少し議論内容から離れているような気がしますが、今の腕先ですと芝生に引っかかる可能性があり、それですと起き上がりに支障をきたすおそれがあるので、3号機のように腕先の面積を増やし丸めた方が良いかと思われます。

写真は3号機の腕先です。
[image]