citbrains/OpenPlatform #31: Gankenkunの破損報告

takayan660, 2017-11-25:
Gankenkunのモーションを作成しようとしたところハードウェアの破損を見つけたので報告します。

* 膝部分でガラスエポキシのリンクが干渉している
 干渉していることにより手で回転させると他より大きな抵抗を感じます。
 左右両足で内側, 外側の計4箇所で発生しています。
[image]

* 股のロール軸サーボ周り
  * サーボの出力軸と反対側の軸が板金と干渉している
[image]
  * サーボの出力軸側の面に固定されている板金の破損
[image]
[image]

takayan660, 2017-11-25:
>・膝部分でガラスエポキシのリンクが干渉している
>3DCAD図
>[image]
>設計ではわずかながらギャップがあります．
>加工精度の問題だと思われます．

>・股のロール軸サーボ周り
>3DCAD図
>[image]

>図面ではギャップがあります．先ほど確認したら部品（ワッシャ）が不足していました．

takayan660, 2017-11-25:
>ガラエポの問題について対応しました.すべての箇所図のように削りました.
>[image]

SatoshiShimada, 2017-11-25:
https://github.com/citbrains/citbrains_humanoid/issues/816 こちらで議論したものを引継いでいます。
以降はこちらで議論しましょう。

shutotakano, 2017-11-27:
股のロール軸サーボ周りについて
不足していたワッシャーを加えました.
間隔があいたことを確認しました.
[image]

yasuohayashibara, 2017-11-27:
問題の確認と対策ができたと思われますので，こちらのissueを閉じます．

closed 2017-11-27 by yasuohayashibara

shutotakano, 2018-05-05:
13番サーボ(1170-A)がしゃがんている状態で180°命令をして発熱,破損.
長穴機構のネジが緩み,ガラエポと干渉,サーボが大きな抵抗を受け発熱,破損.
写真は後者の破損のものです.
[image]
[image]